LubomirJagos/FreeCAD-OpenEMS-Export #1: Please add a little more elaboration and screenshots to README.md

luzpaz, 2020-10-10:
Just so folks have a better understanding of what this project does.

LubomirJagos, 2020-10-11:
Hi, thanks for your interests, I know that I wasn't adding too much description, in short my idea is to program simple python GUI module which you run from FreeCAD and it shows you simple list where you define material parameters for object in FreeCAD and simple grid parameters and it will generate you Octave script which just simple run. I want to do this to play with antenna designs and filters on PCB. I already know how to program GUI in python for FreeCAD and I want to create this week first prototype to play with simulations, I love it, I studied it at university and now working as test engineer for data satellite units for planes.
This is my hobby project and I like to see that other people are interested and it's not just me. Really sorry I would like to regulary continue on this, but sometimes I'm really lazy.

luzpaz, 2020-10-11:
Cool thanks for the reply!. Sounds like an awesome project.  
Having a better README will help with captivating users to your idea :wink:  
FreeCAD motto: "It's done when it's done" :smirk:

luzpaz, 2022-02-27:
bumpity bump on improving the README :stuck_out_tongue_winking_eye:

aster1cs, 2022-10-24:
I'll second that. This is an amazing project with a potentially big impact in the community. (Using StepUp to convert KiCAD to FreeCAD and subsequently FreeCAD-OpenEMS-Export would be a great toolchain to enable patch antenna simulations under consideration of a whole board design.)
However, for new users a slightly more detailed tutorial or a working freecad example project would be really helpful to get started.

MisterHW, 2023-04-29:
A good readme is absolutely necessary, but it's also worth thinking about what to split off into separate tutorials.
For example, I creates this write-up as I was getting into FDTD and FreeCAD-OpenEMS-Export - but by now I know a bit more about the fundamentals as well as intricacies:

https://github.com/MisterHW/Notes/blob/master/FreeCAD_OpenEMS/stub_filter_problem_setup.md

The primary reason I haven't advertised it or provided it as part of the documentation for this repository is that basically all the screenshots are outdated and so much has been improved along the way that it's essentially time to write it from scratch.

In addition to that, some of the challenges I encountered with mesh generation are now being worked on. This mostly concerns
- removal or other treatment of duplicate or closely spaced mesh lines
- generation of smooth mesh transitions

and later probably
- detection of other mesh flaws that need debugging (e.g. automatically flagging issues with insufficient numbers of PML or PMLs that eat into the geometry
- open-circuit detection where conductive objects end up with zero mesh lines in them, which currently has to be diagnosed manually by inspecting the PEC dump

luzpaz, 2023-05-02:
@LubomirJagos  :point_up_2:

luzpaz, 2023-05-07:
@LubomirJagos ping

Basilisvirus, 2024-10-26:
@luzpaz  https://youtu.be/VcJqhsbzR3c?si=5bYWabDL_8adizKM this might help?