LubomirJagos/FreeCAD-OpenEMS-Export #4: BUG?! in file _experimental_GUI_2.FCMacro

eaea74, 2021-03-12:
Hello,

on Win10 with a fresh FreeCAD 0.18 installation the the macro

_experimental_GUI_2.FCMacro

stops creating simulation files at line 715

714			#FOR WHOLE SIMULATION THERE IS JUST ONE EXCITATION DEFINED, so first is taken!
715			item = excitationCategory[0].child(1)

Maybe this has to be changed to:

715			item = excitationCategory[0].child(0)

(this worked for me...)

Greetings

LubomirJagos, 2021-03-28:
Hi, I just finally get some time to continue on this project and make it running, yes you are true, thanks for help. The reason why there was ...child(1) was before I had in each category item "Default ..." so I was skipping it and taking user defined excitation. Later I decided this wouldn't be there and forget to change this.

luzpaz, 2021-04-06:
is the fix implemented yet?

LubomirJagos, 2021-04-06:
Hi, I was recently adding feature to define object priority list and during that fixed this issue.

changed to ...excitationCategory[0].child(0)

closed 2021-04-06 by LubomirJagos