LubomirJagos/FreeCAD-OpenEMS-Export #11: introduce settings format type

MisterHW, 2022-11-12:
To be able to load any settings files, the settings file structure must meet the format implied in the loading routine.
In its current form, no file format information is given, and the loading routine is made robust (to some extent) by providing plausible defaults for missing settings data.

Format information is needed to identify compatible files or take appropriate import actions as the development progresses.

LubomirJagos, 2023-04-27:
I added file into develop branch IniValidator0v1.py which contains ini file field description with values datatypes, during loading it checks file and report errors found, this way file format could be tracked, if it pass through validator (can also through more of them like 0v1, 0v2, 0v7) then it's valid file, so version of GUI will be tight to some versions of validators.

LubomirJagos, 2023-04-29:
In general there is no schema validation for .ini file as for .xml, so it must be implemented by myself. This is example from IniValidator0v1.py where is expected .ini file structure defined and then there is function to check file and give back report about missing mandatory fields or fields with wrong value. It's just formal .ini file check, checking fields and datatype of values, but it's not checking if values are right in term of simulation.

```python
    IniFileSchema = {
        'topLevelGroups': [
            {
                'name': r"MATERIAL-(.+)",
                'mandatory': False,
                'items': [
                    {
                        'name': 'type',
                        'mandatory': True,
                        'allowedValues': r"(metal|userdefined|conducting sheet)"
                    },
                    {
                        'name': 'material_epsilon',
                        'mandatory': 'settings.value("type") == "userdefined"',
                        'allowedValues': 'float'
                    },
                    {
                        'name': 'material_mue',
                        'mandatory': 'settings.value("type") == "userdefined"',
                        'allowedValues': 'float'
                    },
                    {
                        'name': 'material_kappa',
                        'mandatory': 'settings.value("type") == "userdefined"',
                        'allowedValues': 'float'
                    },
                    {
                        'name': 'material_sigma',
                        'mandatory': 'settings.value("type") == "userdefined"',
                        'allowedValues': 'float'
                    },
                    {
                        'name': 'conductingSheetThicknessValue',
                        'mandatory': 'settings.value("type") == "conducting sheet"',
                        'allowedValues': 'float'
                    },
                    {
                        'name': 'conductingSheetThicknessUnits',
                        'mandatory': 'settings.value("type") == "conducting sheet"',
                        'allowedValues': r"(pm|nm|um|mm|cm|m|km)"
                    },
                    {
                        'name': 'conductingSheetConductivity',
                        'mandatory': 'settings.value("type") == "conducting sheet"',
                        'allowedValues': 'float'
                    },
                ]
            },
            {
                'name': r"GRID-(.+)",
                'mandatory': False,
                'items': [
                    {
                        'name': 'coordsType',
                        'mandatory': True,
                        'allowedValues': r"(rectangular|cylindrical)"
                    },
                    {
                        'name': 'type',
                        'mandatory': True,
                        'allowedValues': r"(Fixed Distance|Fixed Count|Smooth Mesh)"
                    },
                    {
                        'name': 'generateLinesInside',
                        'mandatory': True,
                        'allowedValues': "bool"
                    },
                    {
                        'name': 'topPriorityLines',
                        'mandatory': True,
                        'allowedValues': "bool"
                    },
                    {
                        'name': 'units',
                        'mandatory': True,
                        'allowedValues': r"(pm|nm|um|mm|cm|m|km)"
                    },
                    {
```