LubomirJagos/FreeCAD-OpenEMS-Export #40: Please put installation info in README

stevegt, 2024-04-18:
Can someone who knows how to install and use this plugin please send @LubomirJagos a pull request to add installation instructions to the README?  

I found this project via https://github.com/thliebig/openEMS-Project/discussions/119.  It looks real promising, but I use OpenSCAD and sdfx more than FreeCAD these days, so can't figure out where this goes, and I don't want to break my FreeCAD config while trying to figure it out.  

Even just a hint would help -- does this get cloned into the macros directory, or run as a standalone python script, or something else?  I can't find any installation notes anywhere on the Internet, not even in https://github.com/LubomirJagos/FreeCAD-OpenEMS-Export-Plugin-Documentation.  I see other folks on 
(hackaday)[https://hackaday.io/project/176586-freecad-to-openems-plugin] having the same problem.

stevegt, 2024-04-18:
In case it helps anyone, I've found hints in https://github.com/LubomirJagos/FreeCAD-OpenEMS-Export-Plugin-Documentation/blob/main/_text_manuals/FreeCAD%20to%20OpenEMS%20Export%20Plugin%20Tutorial%20-%20Planar%20Filter%20Design.pdf, way down on page 16, in this section:

**Setting openEMS simulation using FreeCAD plugin**

> Run FreeCAD to OpenEMS Export plugin, click on running macros, setup path to folder with
> downloaded plugin and execute macro.

That sentence and the screenshot on that page don't make sense.  Are there two things -- a plugin and a macro, and you use the plugin to execute the macro?  After poking around for a while, I took a wild guess that this is the right procedure:

1. Clone the repo into `~/.local/share/FreeCAD/Macros` 
2. In FreeCAD, go to `Macro | Macros ...`
3. Set the `User macros location` path to where you cloned the repo
4. Choose the `Export for OpenEMS.FCMacro` file
5. Hit `Execute`

But there is no `Export for OpenEMS.FCMacro` file in the repo.  The closest thing seems to be `FreeCAD-OpenEMS-Export.FCMacro`.  So I tried executing that file from the FreeCAD "Macro | Macros ..." dialog.  It throws this error:
```
<class 'FileNotFoundError'>: [Errno 2] No such file or directory: '/home/stevegt/.local/share/FreeCAD/Macro/FreeCAD-OpenEMS-Export/FreeCAD-OpenEMS-Export/ExportOpenEMSDialog.py'
```

It looks like that tried to execute`ExportOpenEMSDialog.py` but for some reason duplicated the parent directory in the path.  So I tried executing `ExportOpenEMSDialog.py` directly, also via the FreeCAD "Macro | Macros ..." dialog.  That does bring up what looks like the export tool's dialog.  I've spent time writing this so haven't gone further than that yet, so don't know if what I'm doing now is the way you're supposed to start this thing or not.

LubomirJagos42, 2024-04-18:
Hi, yes ExportOpenEMSDialog.py is GUI frontend, that .FMacro file is just startup for it meaning to copy it to parent directory, reason why it is so is that macro files are in one folder and this export tool is using more files so you clone it into your macro directory, copy startup file between your macro files and you can use also other macros without changing directory all the time
also I made video about installation https://www.youtube.com/watch?v=OR-79dCMdyo