LubomirJagos/FreeCAD-OpenEMS-Export #41: Error when opening settings

GMellar, 2024-04-18:
When I save the settings and open them again I get the python error "AttributeError: 'NoneType' object has no attribute 'topPriorityLines'". Grids are imported but all associations to grids and materials are lost. 

I'm using the Version from the main branch with FreeCAD Version 0.21.2.

LubomirJagos42, 2024-05-07:
Hi, sorry for late answer, I corrected this, during loading settings .ini file there is delete of all items in UI and then it's repopulate, I tried your scenario on my PC and got this error in report view but loading process was not affected, anyway I repaired it and put it into new release version 0.4.2 please download it from here due I lost access to this repository and I am using new one:
https://github.com/LubomirJagos42/FreeCAD-OpenEMS-Export/releases

LubomirJagos42, 2024-05-07:
Also I recommend to use FreeCAD version 0.20.2 because in latest verison I found error during STL export that distances are twice bigger as set in UI, I reported it to FreeCAD team and tried pre-release version where it's running as it should be, but still current version 0.21.2 on Windows have this error, I developed my GUI addon using 0.20.2 and there is everything running OK.

https://github.com/FreeCAD/FreeCAD/releases