LubomirJagos/FreeCAD-OpenEMS-Export #43: Issue on Linux

EggertEnjoyer123, 2024-06-21:
Hi,
I was unable to run the macro on Linux. I ended up having to change all instances of PySide2 to PySide in every file to get it to work, since the one bundled with FreeCAD is called that.

https://wiki.freecad.org/PySide

"This module just imports the necessary classes from PySide2, and places them in the PySide namespace. This means that in most cases the same code can be used with both Qt4 and Qt5, as long as we use the single PySide module."

LubomirJagos42, 2024-06-21:
Hi, glad that you were able to solve it. I developed and used this on Windows FreeCAD 0.20 and haven't had this issue, can you provide info about your FreeCAD version please? I'll try it in virtualbox to see what's different in Linux FreeCAD, I'm not sure maybe I installed PySide2 into it.

Generally PySide is quite old even PySide2 is starts to be obsolete, what I saw there is PySide6 now used (numbering was changed PySide6 is for Qt6) depends which version of Qt is FreeCAD now using.

If you provide more info about your FreeCAD I can check it.

EggertEnjoyer123, 2024-06-21:
I am on the following:
[code]
OS: EndeavourOS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: makepkg
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.12.3, Qt 6.7.0, Coin 4.0.2, Vtk 9.3.0, OCC 7.7.2
Locale: English/United States (en_US)
[/code]

I used the following command to fix all Python scripts: `find . -name "*.py"| xargs sed -i 's/PySide2/PySide/g'`
I also had to move the FCMacro file up a directory to get it to work properly.

Interestingly, even if I install PySide2, FreeCAD still doesn't let me import it in the Python console. I can only import PySide.

EggertEnjoyer123, 2024-06-22:
I have a few other questions.

I took the filter design from the other repository here (https://github.com/LubomirJagos/FreeCAD-OpenEMS-Export-Plugin-Documentation/tree/main) and I'm getting weird results. Here's what it looks like:
https://i.imgur.com/kUoX2o3.png
I also can't run any generated Octave scripts - it fails with error:
```
error: parse error near line 171 of file /home/eggert/FreeCAD-OpenEMS-Export-Plugin-Documentation/LowPass Filter/LowPassFilter_basic_openEMS_simulation/LowPassFilter_draw_S11.m

  syntax error
mesh.z = np.concatenate((mesh.z, linspace(zmin, zmax-0.3, 3)))
```

I just cloned both repositories and ran the program with no changes. Did I use the wrong version?

LubomirJagos42, 2024-06-22:
about that error, now when I looked on it, I put there for "Grid Settings" > "substrate Z 3 lines" user specified gridlines defined like this:
```python
mesh.z = np.concatenate((mesh.z, linspace(zmin, zmax-0.3, 3)))
```
that's for python script file, for octave it must be updated to
```matlab
mesh.z = [mesh.z, linspace(zmin, zmax-0.3, 3)]
```

I updated examples while ago and probably forgot about this, now at my PC I have this setting split to 2 files, LowPassFilter_basic__octave and LowPassFilter_basic__python which I will upload, they just differ in this since user defined gridlines are chunks of code which are directly inserted into generated script.

To the result image that's wrong, it shoul looks like this
[image]

EggertEnjoyer123, 2024-06-25:
I found out the reason, for some reason your program isn't generating the ground plane correctly.
https://i.imgur.com/hALnaPO.png

LubomirJagos42, 2024-06-25:
yes, userDefined gridlines are chunks of code takes as are and placed into file, so if you put there python code everytime it will be python, if you put octave everytime it will be octave, it's treated just like text insertion, kind a there is no warning
this way you can put any code into generated script

I can't open that .png image, there is error it's broken, looking on that FreeCAD 0.21 issue I think it's ont good to use FreeCAD 0.21, I also reported oen issue with STL export as there was problem, I recommend you should use FreeCAD 0.20, I developed this addon and haven't any problem with it till now, seems like 0.21 is buggy probably 0.22 would be better

Just install FreeCAD 0.20 and you will have no issue