LubomirJagos/FreeCAD-OpenEMS-Export #45: FreeCAD air cube is not at the same location after exporting using this macro

Basilisvirus, 2024-10-26:
When I use the FreeCAD-OpenEMS-Export macro in FreeCAD and my "openEMS_simulation" is created:  
[image]  
And then I open the .py file to run the simulation, I see that the air cube I made for the simulation is not as seen in FreeCAD:  
[image]  

My versions of software are:  
-Python 3.11.9  
-Matplotlib Version: 3.9.2  
-h5py: Version: 3.12.1  
-KiCad: Version: 8.0.5, release build
-FreeCAD: V0.21.2  
[image]  
-I installed the cp311 .whl files:  
[image]  

I do not know how to debug this further. I attach the FreeCAD, macro settings and macro export files here. 
[simulationWaveProp.zip](https://github.com/user-attachments/files/17530990/simulationWaveProp.zip)

LubomirJagos42, 2024-10-26:
some months ago I saw this problem and I think it was already repaired in FreeCAD, please try to use older version 0.20.x
in this 0.21.2 there is something wrong with STL export like position and dimension are not right exported

Basilisvirus, 2024-10-26:
> some months ago I saw this problem and I think it was already repaired in FreeCAD, please try to use older version 0.20.x in this 0.21.2 there is something wrong with STL export like position and dimension are not right exported

You were right. Works with FreeCAD 0.20.2  

[image]  

So this bug is it a known issue? like its listen in their github's issues?

closed 2024-10-27 by Basilisvirus

LubomirJagos42, 2024-10-29:
hmm, incompatibility what I'm sure will not run is to use Python 3.11, don't try 3.12, few people tried that and it's not running since CSXCAD and openEMS interface are cython code which is compiled into DLL and there is compatibility table somewhere, in short now openEMS and CSXCAD are created for python 3.11 and it's not possible to run it on 3.12
maybe if you build openEMS manualy you can make it running but at the moment out of box libs are for python 3.11, I think 3.10 is also ok

their names are:
CSXCAD.cp311-win_amd64.pyd
openEMS.cp311-win_amd64.pyd